open-ephys/plugin-GUI #626: Could the Record Control processor have a buffer option?

ChucklesOnGitHub, 2024-09-10:
User was asking to have the option to append some seconds of buffered data to file when starting recordings via TTL with the Record Control processor.

jsiegle, 2024-09-10:
This would require a pretty big overhaul of the recording architecture, so probably won't happen. Is there a reason they can't just record continuously and extract the relevant segments offline?

gomezramirezm, 2024-09-12:
Hello, I just installed the version of Open Ephys in a new machine in the lab running Neuropixels. I couldn't find the "Record Control" Plugin in the Installer. Has this option been removed, or am I doing something wrong? See picture below of plugin installer options sorted alphabetically. Thank you! 

[image]

anjaldoshi, 2024-09-12:
Hi @gomezramirezm,

The Record Control plugin comes pre-installed with the GUI. It is available under "Utilities" in the Processor List.

[image]

ChucklesOnGitHub, 2024-10-11:
This is commonly used for experiments in which singing behavior of the animals is monitored over hours and recordings are only triggered when song starts using an amplitude threshold to avoid long files. The buffer would be useful to get a few secs of ephys before sound onset instead of clipping at the timestamp of the threshold cross