open-ephys/plugin-GUI #627: Loading timestamp for electrode in open e-phys

cellywiththesticks, 2024-09-13:
Hi all,

I am trying to see what my timestamps looks like in open e-phys for electrode 10. My timestamp files is currently saved as an .npy . Any help will be greatly appreciated!

jsiegle, 2024-09-16:
I'd recommend either using [`open-ephys-python-tools`](https://github.com/open-ephys/open-ephys-python-tools) or [`open-ephys-matlab-tools`](https://github.com/open-ephys/open-ephys-matlab-tools) to load the entire recording. That will automatically load the timestamps for each data stream.

You can also replay the recording in the GUI by dragging the `structure.oebin` file into the [File Reader](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/File-Reader.html).

jsiegle, 2024-09-17:
What type of device did you use to record your data (e.g. Open Ephys Acquisition Board)? Depending on what you used, you may not need to do any additional synchronization.

cellywiththesticks, 2024-09-17:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Yes I used the acquisition board but my timestamps start at around 2000 seconds which makes no sense since my recording ends at 25 min

jsiegle, 2024-09-17:
The timestamps are always relative to the time you started streaming data, not the time when you started recording. This is so if you start and stop recording multiple times while data continues to stream, all of the timestamps will be aligned to the same start time (i.e., you will know exactly how much time has elapsed between recordings). If you stop data streaming by pressing the "play" button, then the timestamps will reset to 0.

If the recording ends at 25 minutes, the last timestamp should be around 25 * 60 + 2000 = 3500 seconds. Is that what you see?

chiyu1203, 2025-04-14:
Hi @cellywiththesticks,
I would like to quickly follow up this discussion since I bumped into the same issue last week. The recording was 37 mins long but perhaps I started to stream the data much earlier (20 mins earlier), the timestamp was count around 20 mins earlier. Is there any post-hoc way to precisely adjust the timestamp so that it is align to the start of the recording not the start of the streaming?

jsiegle, 2025-04-14:
Hi @chiyu1203 – can you just subtract the first recorded timestamp from all of the others? That should re-align the subsequent timestamps with the start of the recording, rather than the start of acquisition.

closed 2025-06-08 by jsiegle