open-ephys/plugin-GUI #628: probe map _ older versions of open_ephys ( 0.5 )

Aryo-Zare, 2024-09-16:
I have some older neuropixels data that I would like to revive. I guess they were recorded by open-ephys GUI version 0.5… .
I was naiive at that time & did not export selected electrode contact locations for each channel ( μm from the tip ) as probeInterface JSON files via the npx plugin in GUI.
In the newer versions of the GUI these electrode contact location information are internally saved in the recording files ( I guess in setings.xml  or  structure.oebin ).
However, I’m not sure about the older versions of the GUI.

I attached two related files. I converted both files to json, as github did not support .xml & .oebin.
[settings.xml.json](https://github.com/user-attachments/files/17014973/settings.xml.json)
[structure.oebin.json](https://github.com/user-attachments/files/17014974/structure.oebin.json)
Would you please tell me if there is any probe-map or similar electrode location information for each channel, present in these files ?

jsiegle, 2024-09-16:
The `settings.xml` file has information about the selected bank for each channel (e.g. `CH0="0"`). You can use that to figure out the electrode locations in microns as follows:

```python
channel = 12 # number between 0 and 383
bank = 1 # 0, 1, or 2
xpositions = [27, 59, 11, 43]
xpos = xpositions[channel % 4]
ypos = ((channel + 384 * bank) - channel % 2) * 10 
```

This assume you're using a 10 mm NHP probe with staggered site layout.

Aryo-Zare, 2024-09-17:
That’s a relief !
Thanks a lot for your comlete & swift response.

Aryo-Zare, 2024-09-25:
My I still ask a question ?
I could extract the individual channel locations.
However, in GUI version 0.5.5.3, there is always one channel’s ‘bank’ information missing : this channel is always channel number 191 !!  I tested this in various experiments.
In GUI version 0.6.4, this bank information is also included, for all channels with none of them missing. 

How can I find the bank information for channel 191, & why is always this specific channels’ info missing ?

jsiegle, 2024-09-25:
Channel 191 is always the internal reference. It doesn't record any signals, so the bank does not matter.