open-ephys/plugin-GUI #632: Crash when opening recordings in OpenEphys format

matteo-filippini, 2024-10-11:
Hi, I'm trying to open recordings I made in .openephys format from the GUI through the source file reader. As soon as I load the files, the GUI crashes with the attached log. A small portion of the recordings open without any issues, but unfortunately, most of them don't. Do you have any idea how I can figure out the problem? I later switched to the DAT format because, with more than 128 channels, the OpenEphys format crashes when starting the recording, but it’s important for me to reopen these data. Thanks!
[configs-api8.zip](https://github.com/user-attachments/files/17341678/configs-api8.zip)

medengineer, 2024-10-11:
Hi,

Can you send a link to one of the recordings causing the crash? From the attache log it looks like some files may have been manually renamed?

matteo-filippini, 2024-10-11:
I thought the same, but I double-checked, and the file names match those in structure.openephys. The remaining options are either corrupted data (but I can open them as binary files using Python without any issues) or a filesystem limitation (which seems unlikely) when trying to open all those files at once. To test this, I tried editing structure.openephys, leaving only one file, but it still crashes. I’m out of ideas! Thanks for any tips. I’ve attached the WeTransfer links (about 5GB). 
cheers
https://we.tl/t-mgU92JPESo
https://we.tl/t-NxWBEqzG5h
https://we.tl/t-qUghHdWS01

oepavel, 2024-10-14:
Hi,

Looks like the links have expired. Can you refresh them and I can take a look today?

Thanks

matteo-filippini, 2024-10-15:
there you go
https://we.tl/t-1X4C32evqQ
https://we.tl/t-340Ci15Xxn
https://we.tl/t-dntGbFzXB7
thanks!

matteo-filippini, 2024-10-22:
Any updates? Have you had a chance to look into it? I don't mean to rush, but the link expires again today, so if you can at least download it, I'd really appreciate it. Thanks for anything you're able to do
cheers

medengineer, 2024-10-22:
Yes, I was able to download and reproduce the crash during loading. Will follow up with a fix here shortly.

medengineer, 2024-10-24:
I was able to get the data to load in FileReader by modifying the <RECORDING number> tag value in the structure.openephys file. In the example you sent above, I changed the value from 13 to 1. 

Assuming this is the same issue for the rest of the recordings, ensuring the recording number is set to 1 in each recording's structure.openephys file should allow you to load the files in the GUI's FileReader.

matteo-filippini, 2024-11-19:
I tried several recordings that didn't work and editing that recording number solved the problem, thank you very much! cheers