open-ephys/plugin-GUI #633: Duplicate / chunked files in OpenEphys 0.4.2.1.

mgemayel, 2024-11-18:
Hi,

Thanks for making OpenEphys! 

We have inherited a dataset acquired with OpenEphys version 0.4.2.1 using a A1x32 Neuropixel probe (i.e. with 32 channels). 

In one session, multiple experiments (~10) were conducted back to back within a same subject in a period of a few hours.

We've noticed that in certain experiments within a session there are duplications of all files, or duplications and triplications of all files (e.g. 100_CH1.continuous, 100_CH1_2.continuous, 100_CH1_3.continuous). 

In the case of experiments with duplications only, the _2 (duplicated) files are not the same size as the original files.

In the case of experiments with duplications and triplications, the _3 (triplicated) files are the same size as the original files, but the _2 (duplicated) files are not.

Attached is the output for `ls -lah <folder>` for a normal experiment, an experiment with file duplicates only, and an experiment with file duplicates and triplicates.

We would appreciate any insight regarding why these duplications and triplications occur / why the recordings are being chunked in some cases but not in others?

[Normal.txt](https://github.com/user-attachments/files/17799782/Normal.txt)
[Duplicates only.txt](https://github.com/user-attachments/files/17799784/Duplicates.only.txt)
[Duplicates and triplicates.txt](https://github.com/user-attachments/files/17799787/Duplicates.and.triplicates.txt)

medengineer, 2024-11-19:
Hi,

Whenever data acquisition is stopped and restarted between recordings in the GUI (and the root directory does not change), a new file gets created and appended with _<experiment_number>. The experiment number is incremented every time acquisition is started/restarted. 

Therefore, the triplicate example above is actually three different (sequential) experiments where acquisition was stopped and restarted between the recordings. The first and third experiment ran for roughly the same amount of time and the second experiment for much shorter in between the two. There should not be any replicated data among the files.

mgemayel, 2024-11-20:
Hi, 

Thank you for your response. 

We were under the impression that in the Open Ephys Legacy Format (i.e prior to OpenEphys 0.5.0) stopping and restarting data acquisition resulted in the creation of a new **folder**,  which was appended with   _<experiment_number>. However, what we observed are multiple versions of all files **within** one folder, distinguished by underscores (_2, _3). 
Our experimenter does not recall stopping and restarting the data acquisition within one recording, but since it has been a while, this is a distinct possibility. 

If our understanding of the legacy format is incorrect, we apologize for the confusion. In that case, please feel free to close the issue.

jsiegle, 2024-11-21:
Even the latest version of the [Open Ephys format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Open-Ephys-format.html) does not create new directories for each experiment/recording. This format was intended to have a completely flat directory structure.

The [Binary format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html) (now the default format, which is recommended for most use cases) is the one that creates new sub-directories every time recording is stopped and re-started.

mgemayel, 2024-11-21:
Thank you for the clarification and your patience.

closed 2024-11-21 by mgemayel