open-ephys/plugin-GUI #634: Issue synchronizing Neuropixel and NI-DAQ data streams

ellecj729, 2024-11-25:
Hello!

I am trying to synchronize multiple analog inputs from a NI-DAQ (NI USB-6211) with the data stream from a Neuropixel 2.0 probe. I am trying to use the Neuropix-PXI as the main clock. As recommended the SMA is set to output and the cable is connected to DI0 on the NI-DAQ. I am using a merger node with Neuropix as input A and NI-DAQ as input B and have the 1st digital channel set as the sync line for the NI-DAQ in the record node. With this configuration, once I start acquisition, I am able to read the analog inputs accurately through the LFP viewer and the probe is synchronized but the NI-DAQ sync line never turns green. 

I have confirmed through NI-MAX that the DAQ is receiving the 1Hz signal from the basestation and followed the recommendations in the synchronization tutorial but so far nothing has resolved the issue. Any advice would be greatly appreciated!

ellecj729, 2024-11-25:
Thanks for the reply!! 

I tried lowering the sample rate to 1kHz but have the same issue. Also I should clarify that I don't see the digital 1Hz signal (at least the event overlay) in the LFP viewer.

medengineer, 2024-11-25:
Ah okay, that's a separate problem then. So you can see the NIDAQ analog inputs in the LFPViewer but not the digital 1Hz event overlays.

Is there any error message in Debug console of the GUI? Does changing the digital input connection on the NIDAQ device make any difference, (ex: from line 1 to line 2)?

closed 2025-06-09 by medengineer