open-ephys/plugin-GUI #637: Question: Will update firmware for open-ephys affect the proper use of spikeGLX?

DaohanZhang, 2024-12-05:
[image]
Hey, guys! I'm new to open-ephys but i also used spikeGLX for recording neuropixels. But my neuropixels plugin in open-ephys ask me to update firmware version of Basestation connect board from 3.2189 to 3.2191. How do I perform this update (I mean I need to upload what files?) Also, will this update made me fail to record using spikeGLX?

jsiegle, 2024-12-06:
What version of SpikeGLX are you using? The latest release (v20190413-phase3B1) also requires BSC firmware 3.2191. If you upgrade to that version, you can use SpikeGLX and Open Ephys side by side with the same firmware.

jsiegle, 2025-06-08:
It's still true that the latest version of the Neuropix-PXI plugin requires the same basestation firmware at the latest version of SpikeGLX.

closed 2025-06-08 by jsiegle