open-ephys/plugin-GUI #640: juce_OwnedArray.h

kevkdev, 2024-12-09:
I'm posting this here since the plugin-GUI github has a copy of the JUCE library (instead of pointing to it via a sub-module, which you may want to consider to do some day).

On line 417 of _juce_ownedArray.h_ (in _set_()  method)

`if (toDelete.get() == newObject)`

should be

`if (toDelete.get() != newObject)`

jsiegle, 2024-12-09:
We can't include the standard JUCE library directly, because we had to customize some functions (e.g., to be able to send spike through the signal chain as MIDI events).

Can you explain the rationale for this change? This is a critical part of the JUCE core, and I'd be surprised if they made a mistake here.

kevkdev, 2024-12-10:
With more context:
`   ObjectClass* set (int indexToChange, ObjectClass* newObject, bool deleteOldElement = true)
    {
        if (indexToChange >= 0)
        {
            std::unique_ptr<ObjectClass> toDelete;

            {
                const ScopedLockType lock (getLock());

                if (indexToChange < values.size())
                {
                    if (deleteOldElement)
                    {
                        toDelete.reset (values[indexToChange]);

                        if (toDelete.get() == newObject)
                            toDelete.release();
                    }

                    values[indexToChange] = newObject;
                }
                else
                {
                    values.add (newObject);
                }
            }
        }`
        
        The if statement to check for deletion compares the new pointer vs the old. if they are the same, then delete - which logically doesn't make sense. They should be different, and if so, then delete the old.

jsiegle, 2024-12-10:
`unique_ptr::release()` specifically does *not* call the destructor, so this is preventing deletion in case the element is being replaced with itself.

If the pointer is not released, the destructor will be called when the `unique_ptr` goes out of scope.

kevkdev, 2024-12-11:
Yeah, I see that now. As I was working on reducing the danging pointers in my code, I saw that and it appeared to resolve some of it; undoing this change didn't make a difference in JUCE dangling pointer complaints. :man_shrugging: Closing.

closed 2024-12-11 by kevkdev