open-ephys/plugin-GUI #642: Error in storing TTL events (recordings in OpenEphys format)

AnnaB27, 2025-01-06:
Hi all,
I am a Phd student who works at the university of Parma. Up until now, We conducted several experiments using the OE GUI for recordings. We had an Opal Kelly acquisition board and we were using version 0.4 of the guide. Recently, the board broke and we had to fix it and replace it with the most recent acquisition boards that use the new Open Ephys FPGA (instead of Opal Kelly). We encoutered two issues:
1. We tried to use the guide but the old version 0.4 couldn't detect the new acquisition board. Is there a way?
2. we then downloaded the newest version of the guide 0.67 (We usually record in openephys format, so i installed the necessary plug-in as explained in the docs) and now we could read the acquisition board, but there is another problem. 
I tried and performed a "fake" acquisition of an experiment, and everything seemed fine, until I realized that there is a problem with the Task Events: during the recording online, I can see that OE receives the different TTLs from the behavioural software that controls the task, but once the data is saved, and I try to open the .EVENT file, an error appear about the headers file. I opened as text the file and some informations, such as sample rate, buffersize,bitvolts they are nowhere to be found. I double-checked the signal chain to be sure everything was in place. Since I am not an expert, am I missing something? 
I hope I was clear enough, thank you in advance.

medengineer, 2025-01-07:
Hello,

There have been significant changes since GUI version 0.4, including OpenEphys format recorded files. If you are using custom scripts or tools specific to the 0.4 version, you will need to update them to support these changes. 

There are official packages in [Python](https://github.com/open-ephys/open-ephys-python-tools) and [MATLAB](https://github.com/open-ephys/open-ephys-matlab-tools) to load data from Open Ephys GUI in all supported formats. We suggest using them as documented, otherwise you can update any custom scripts based on the code in those packages.

Specifically, here is how Open Ephys format event files are loaded in [Python](https://github.com/open-ephys/open-ephys-python-tools/blob/main/src/open_ephys/analysis/formats/helpers/oe_fast_loader.py#L193) and [MATLAB](https://github.com/open-ephys/open-ephys-matlab-tools/blob/main/open_ephys/analysis/formats/OpenEphysRecording/OpenEphysRecording.m#L195).

AnnaB27, 2025-01-08:
Hi,
Thank you! and yes we also use customized scripts for data extraction and analysis, but that is another issue that we are handling, as you said, by updating our custom scripts for the latest version. 
My question was referring to a different tool. I was simply trying to open the data on NeuroExplorer (v.5, latest version) for data visualization and it is able to load the continuous files for all channels but not the .events file to visualize all the events where it gives me an error for Invalid header.file, it seems as it "doesn't like" how events are stored or there are missing information for some reason.

medengineer, 2025-01-08:
I see, NeuroExplorer uses all the fields in the v0.4 OpenEphys format event header files, some of which we removed in v0.5+ because they were redundant w/ the corresponding stream info. I just confirmed that manually adding back those fields loads 0.6 files correctly in NeuroExplorer.

The temp fix is to copy the 3 missing event header lines (i.e. sampleRate, bufferSize, bitVolts) from a 0.4 events file to each of your 0.6 event files. Not ideal, but a script to do this automatically should be straightforward, happy to help w/ that. 

Given that removing this metadata is causing issues in downstream software I think it's worth restoring in 0.6+ and pushing out a new version of the plugin. Will update this issue when that's ready.

AnnaB27, 2025-01-09:
I understand, that's good to know, at least it's not an issue related directly to the GUI installation or something. It's not like we use stricktly NEX in our analysis pipeline but it was still always very useful as first step, for quickly visualize our recordings to check for mistakes or problems, so if you are able to work on restoring this feuture in a new plugin that would be great! Meanwhile we will use the temporary fix you suggested, since it still works
Thank you very much!