open-ephys/plugin-GUI #643: recording stops

gongsjanelia, 2025-01-13:
When using the old Open Ephys Acquisition Board (version v2.2, revision 3, January 2017) with GUI version 0.6.7 to record data at 30 kS/s, the system records fine for about one minute, but then all data (AUX, ADC, and spike data) stops appearing in the GUI. No error messages are displayed. What could be causing this issue, and how can it be resolved

ZXHzxh0125, 2025-05-14:
Is your problem solved?I've encountered a similar problem.When I use the Open Ephys Acquisition Board (version v2.2, revision 3, January 2017) with GUI version 0.6.7 to record 128 channels of neural signals, about 3 minutes into the recording the software starts to lag with a read failure 32 error.my signal chain also is simple OE FPGA acq board --bandpass filter--record node --LFP viwer

jsiegle, 2025-05-16:
Hi @ZXHzxh0125  -- answers to a few questions will help us troubleshoot: 
- Do you also have an internal SSD for writing data?
- Is the acq board connected to a USB 3.0 port?
- Does the problem occur after you start acquisition, or only after starting recording?
- Are there any other programs running on your computer?