open-ephys/plugin-GUI #644: Open_ephys version 0.6.7 cannot connect to the basestation on a windows 11 machine.

Aryo-Zare, 2025-01-15:
I had used open-ephys version 0.6.4 before without problem. This was installed in a windows 10 computer.
Recently I installed v-0.6.7 on a windows 11 machine.
It cannot connect to the basestation.

[image]

In contrast to the error message in the GUI “no basestations detected” the Log (visible at the side of GUI in the screenshot) mentions detecting the basestation but cannot open it !
I had installed the enclustra drivers without any problem. The NI device is working properly according to windows & its lights are green.

Any idea about the underlying issue ?

medengineer, 2025-01-15:
Assuming you installed via PluginInstaller can you post a screenshot of the installed version from the PluginInstaller? You likely need a basestation firmware update.

https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Neuropixels-PXI.html#updating-basestation-firmware

Aryo-Zare, 2025-01-16:
Thanks a lot ! It’s resolved !
The info section of the neuropixels plugin showed nothing. Hence, I needed to first downgrade it to a lower version ( v0.6.4 ) for it to appear.
Then I accessed the firmware update section & uploaded the files. Subsequently, I upgraded the neuropixels plugin again.