open-ephys/plugin-GUI #646: negative timestamps (all -1)

SusanL82, 2025-02-11:
Hello, 

In our recent tryout recordings with a single neuropixels 1.0 probe, I've been getting wrong timestamps. 
When reading the recordings with the matlab tools, the number of timestamps seems fine, but every timestamp has the value -1. 
What could be the reason for this? And how can we prevent it for future recordings? 

Thanks, 
Susan

medengineer, 2025-02-11:
Hello,

The timestamps values rely on a common synchronization signal to compute synchronized timestamps from the raw sample numbers. When there is no detected synchronization signal, the timestamps all default to -1.

To generate timestamps, you can set your NPX plugin SMA config from INPUT to OUTPUT to generate a 1Hz clk TLL that serves as the synchronization signal. 

With a single 1.0 probe, since the AP and LFP bands are sampled by the same clock on the probe’s headstage, the timestamps file is not necessary to align data streams. You can get the same timestamp values offline by dividing the AP and LFP sample numbers by 30000 and 2500 respectively.

You can find more details re: synchronization [here](https://open-ephys.github.io/gui-docs/Tutorials/Data-Synchronization.html)

SusanL82, 2025-02-13:
Ok, good to know that there is not something broken in our recordings. 

We only have the probe for now, and are working to implement some other data streams. I guess the timestamps will show up when we get those up and running then.

closed 2025-02-13 by SusanL82