open-ephys/plugin-GUI #648: Latest neuropix-PXI plugin (0.7) crashes

chris-angeloni, 2025-02-24:
I recently updated the GUI to the most recent version (0.6.7), upgraded the firmware for the neuropixels basestation and card, and upgraded the Neuropix-PXI plugin to 0.7:

[image]

However, I cannot get the latest version of the plugin to run. Immediately upon adding it to the signal chain the GUI crashes. Downgrading to the 0.6.6 version of the plugin gets rid of this behavior, and probe signals look normal.

Any help is appreciated!

anjaldoshi, 2025-02-24:
Hi @chris-angeloni, could you please confirm if you restarted your computer and performed a power cycle on the PXI chassis after updating the basestation firmware? This step is necessary to ensure that the changes take effect.

Could you also share the crash log file generated during that run with the plugin version 0.7.0? It can be located in the folder `C:\ProgramData\Open Ephys\configs-api8`, and the file name should be `activity_<timestamp>.log` (for instance, `activity_2025-02-24_15-21-22.log`).

chris-angeloni, 2025-02-24:
I did restart and power cycle the chassis. Here is the crash log:

`[open-ephys] Session Start Time: 24 Feb 2025 16:50:28
[open-ephys] Open Ephys GUI v0.6.7 (Plugin API v8)
[open-ephys] JUCE v6.0.8
[open-ephys] Operating System: Windows 10
[open-ephys] CPU: 12th Gen Intel(R) Core(TM) i7-12700KF (20 core)
[open-ephys][debug] Creating processor graph...
[open-ephys][debug] Creating audio component...
[open-ephys] Audio device name: Digital Audio (S/PDIF) (High Definition Audio Device)
[open-ephys] Audio device type: Windows Audio
[open-ephys] Audio output channels: 2
[open-ephys] Audio device sample rate: 44100
[open-ephys] Audio device buffer size: 1024
[open-ephys][debug] Connecting audio component to processor graph...
[open-ephys][debug] Creating UI component...
[open-ephys][debug] Created message center.
[open-ephys][action] Button 'About' clicked
[open-ephys][debug] Created info label.
[open-ephys][debug] Created graph viewer.
[open-ephys][debug] Refreshing state for Info
[open-ephys][debug] Data Viewport adding tab with index 1
[open-ephys][debug] Refreshing state for Info
[open-ephys][debug] Data Viewport adding tab with index 2
[open-ephys][debug] Refreshing state for Graph
[open-ephys][debug] Created data viewport.
[open-ephys][debug] Created editor viewport.
[open-ephys][debug] Created control panel.
[open-ephys][debug] Created filter list.
[open-ephys][debug] Created plugin manager
[open-ephys][debug] Loading Plugin: ArduinoOutput... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: BasicSpikeDisplay... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 2 plugins
[open-ephys][debug] Loading Plugin: ChannelMappingNode... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: CommonAverageRef... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: FilterNode... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: LfpDisplayNode... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: PhaseDetector... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: RecordControl... 
[open-ephys][debug] Adding processor plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: neuropixels-pxi... 
[open-ephys][debug] Adding data thread plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] Loading Plugin: nidaq-plugin... 
[open-ephys][debug] Adding data thread plugin
[open-ephys][debug]   Loaded with 1 plugin
[open-ephys][debug] ProcessorList::fillItemList()
[open-ephys][debug] Processor List - creating item for Merger
[open-ephys][debug] Processor List - creating item for Splitter
[open-ephys][debug] Processor List - creating item for File Reader
[open-ephys][debug] Processor List - creating item for Record Node
[open-ephys][debug] Processor List - creating item for Audio Monitor
[open-ephys][debug] Processor List - creating item for Event Translator
[open-ephys][debug] Processor List - creating item for Arduino Output
[open-ephys][debug] Processor List - creating item for Spike Detector
[... 65 lines cut ...]
14: PluginClass::setPluginData + 0xafbb
15: PluginClass::setPluginData + 0x5df9
16: Visualizer::startCallbacks + 0x116aa
17: juce::UndoManager::perform + 0x44
18: Visualizer::startCallbacks + 0x9bfc
19: Visualizer::startCallbacks + 0xc0d4
20: juce::ComboBox::mouseUp + 0x290
21: juce::LowLevelGraphicsPostScriptRenderer::writeXY + 0x7020
22: juce::Component::internalMouseUp + 0x3ab
23: juce::Button::setButtonText + 0x1f9
24: juce::MouseInputSource::handleEvent + 0x1f8
25: juce::MouseInputSource::handleEvent + 0x96
26: juce::TooltipWindow::displayTip + 0x6db
27: juce::TooltipWindow::displayTip + 0xa0e
28: juce::DrawableShape::pathChanged + 0xa1c
29: juce::MouseInactivityDetector::wakeUp + 0x259
30: DispatchMessageW + 0x741
31: DispatchMessageW + 0x201
32: juce::MessageManager::runDispatchLoop + 0x111
33: juce::JUCEApplicationBase::main + 0xa9
34: TiledButtonGroupManager::setMinPaddingBetweenButtons + 0xea6
35: BaseThreadInitThunk + 0x1d
36: RtlUserThreadStart + 0x28

`

anjaldoshi, 2025-02-24:
Thanks for confirming and sending the crash log. Unfortunately, it doesn’t provide much information about the cause of the crash.  

Did you attempt to add the Neuropix-PXI plugin version 0.7 into the signal chain again after it crashed the first time? Does it crash every time you try to use it? If so, could you please copy the contents of the console after the GUI crashes? This information might provide more details about the location of the crash during the plugin initialization process.

anjaldoshi, 2025-02-25:
That's strange. It might be an issue with the DLL files. Could you please check if the latest Neuropixels library DLLs are installed? Navigate to `C:\ProgramData\Open Ephys\shared-api8` and look for these files: `NeuropixAPI_x64_3_70_2.dll`, `FpgaManager.dll`, and `FTD3XX.dll`.

If they're already there, try this:

1. Launch the GUI and make sure the signal chain is cleared.
2. Open the Plugin Installer, uninstall the Neuropix-PXI plugin, and then close the GUI.
3. Delete the `NeuropixAPI_x64_3_70_2.dll`, `FpgaManager.dll`, and `FTD3XX.dll` files from `C:\ProgramData\Open Ephys\shared-api8`.
4. Relaunch the GUI, reinstall the Neuropix-PXI plugin (v0.7.0), and then try adding the plugin to the signal chain.

Let me know if that resolves the issue.

anjaldoshi, 2025-02-25:
Thank you for going through the debugging steps with me. This is the first time we're encountering this crash, so we want to ensure everything is in order.

If you have some time, the last thing to try is to compile the GUI and the Neuropix-PXI plugin locally in debug mode and run it. This process should pinpoint exactly what's happening and won't interfere with the existing installation of the GUI on your system. 

If you're up for it, here are the instructions:

1. [Compiling the GUI](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-the-GUI.html)
2. [Compiling plugins](https://open-ephys.github.io/gui-docs/Developer-Guide/Compiling-plugins.html)

You can find the source code for the Neuropixels plugin here: [Neuropixels-PXI GitHub repository](https://github.com/open-ephys-plugins/neuropixels-pxi). Please use the code from the `main` branch for both.

Let me know if you get stuck or need help with anything I mentioned above. Debugging it will also be helpful for us in case we need to fix something in the code or if we encounter a similar bug in the future.

chris-angeloni, 2025-02-26:
Hi Anjal, thanks for your help. I will try to compile from source sometime this week or next and let you know how it goes.

paolahydra, 2025-04-01:
Hi @anjaldoshi 

We have a new OneBox that we are trying to get to work, so we need the latest versions of the Neuropixels-PXI plugin.
However, we are experiencing the same issue reported by @chris-angeloni above: we get the same error message using either 0.7.0 or 0.7.1.

It doesn't sound like you guys made any progress with the debugging. Should I give it a try?

Thank you.
Paola

chris-angeloni, 2025-04-01:
I'm in the middle of data collection so I cannot try to compile from source for a while, if you could give it a try that would be awesome

anjaldoshi, 2025-04-01:
Hi @paolahydra and @chris-angeloni

Before trying to compile the plugin on your own, can you install the latest version of the Microsoft Visual C++ Redistributable from [here](https://aka.ms/vs/17/release/vc_redist.x64.exe)? See if that fixes the crash you're seeing when using Neuropix-PXI v0.7.0 or v0.7.1 from the Plugin Installer.

paolahydra, 2025-04-01:
Sure, I'll try this.

paolahydra, 2025-04-02:
Hi Anjal, 
I installed the version of MV C++ Redistributable that you linked.
We made progress, so OE no longer crashes when I try to add the OneBox to the signal chain.

However, I still get this error:

```
[open-ephys] Session Start Time: 2 Apr 2025 9:52:32
[open-ephys] Open Ephys GUI v0.6.7 (Plugin API v8)
[open-ephys] JUCE v6.0.8
[open-ephys] Operating System: Windows 10
[open-ephys] CPU: Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz (8 core)
[open-ephys] Audio device name: Realtek Digital Output (Realtek High Definition Audio)

[open-ephys] Audio device type: Windows Audio
[open-ephys] Audio output channels: 2
[open-ephys] Audio device sample rate: 44100
[open-ephys] Audio device buffer size: 1024

[open-ephys] Loading configuration from C:\ProgramData\Open Ephys\configs-api8\lastConfig.xml
[open-ephys] Creating processor with name: OneBox
[open-ephys] Scanning for devices...
[open-ephys]   Found 1 device.
[open-ephys] Stored OneBox serial number 21470088
[open-ephys] No OneBox found at slot 29572218
[open-ephys]   Could not open OneBox
[open-ephys] Setting slot 29572218 sync as input.
[open-ephys] Failed to clear SM_Output_StatusBit on slot 29572218, error code = 31
[open-ephys] Failed to clear SM_Output_SMA1 on slot 29572218, error code = 31
[open-ephys] Failed to connect SM_Output_StatusBit and SM_Input_SMA1 on slot 29572218, error code = 31
[open-ephys] ***ERROR*** closeBS slot 29572218: NO_SLOT
```

jsiegle, 2025-04-02:
Hi @paolahydra – can you confirm you're using the latest version of the Neuropixels plugin (should be 0.7.1, visible in the Plugin Installer)? If not, please upgrade.

If that doesn't fix the crash, can you let us know if it happens even after closing Open Ephys and power cycling your OneBox?

paolahydra, 2025-04-03:
Hi @jsiegle,
Yes I am using the latest version 0.7.1.
I power cycled the OneBox and restarted openephys.
It read the OneBox on slot 16 first, which was new. But then it lost it again:

```
[open-ephys] Open Ephys GUI v0.6.7 (Plugin API v8)
[open-ephys] JUCE v6.0.8
[open-ephys] Operating System: Windows 10
[open-ephys] CPU: Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz (8 core)
[open-ephys] Audio device name: Realtek Digital Output (Realtek High Definition Audio)

[open-ephys] Audio device type: Windows Audio
[open-ephys] Audio output channels: 2
[open-ephys] Audio device sample rate: 44100
[open-ephys] Audio device buffer size: 1024

[open-ephys] Loading configuration from C:\ProgramData\Open Ephys\configs-api8\lastConfig.xml
[open-ephys] Creating processor with name: OneBox
[open-ephys] Scanning for devices...
[open-ephys]   Found 1 device.
[open-ephys] Stored OneBox serial number 21470088
[open-ephys]   Opened OneBox on slot 16
[open-ephys] Setting slot 16 sync as input.
[open-ephys] OneBox initialized on slot 16
[open-ephys] Creating processor with name: Neuropix-PXI
[open-ephys] Scanning for devices...
[open-ephys]   Found 1 device.
[open-ephys]    Slot 16 did not match desired platform.
[open-ephys] Validating part number: PRB_1_4_0480_1
[open-ephys] NeuropixThread::setMainSync
[open-ephys] Creating processor with name: OneBox
[open-ephys] Scanning for devices...
[open-ephys]   Found 1 device.
[open-ephys] OneBox with serial number 21470088 already connected!
[open-ephys] No OneBox found at slot 29032448
[open-ephys]   Could not open OneBox
[open-ephys] Setting slot 29032448 sync as input.
[open-ephys] Failed to clear SM_Output_StatusBit on slot 29032448, error code = 31
[open-ephys] Failed to clear SM_Output_SMA1 on slot 29032448, error code = 31
[open-ephys] Failed to connect SM_Output_StatusBit and SM_Input_SMA1 on slot 29032448, error code = 31
[open-ephys] ***ERROR*** closeBS slot 29032448: NO_SLOT
```

paolahydra, 2025-04-03:
Hold on, sorry.
There was something weird with the signal chain reloaded at startup.
I started again and it is working fine now! 👍

anjaldoshi, 2025-06-24:
Hi @SaraMederos,

Thanks for the details. I tried reproducing the issue on my end, but everything works fine with the Rhythm Plugins and the TTL Display Panel plugin using NPX plugin v0.7.2.

Are you still experiencing the issue? If so, did you happen to restart your machine after installing the latest Microsoft Visual C++ Redistributable? In some cases, outdated Redistributable files may linger after an update and cause odd behavior