open-ephys/plugin-GUI #652: Unable to use neuropix-PXI 0.7.X plugin using GUI 0.6.7

ziyuw2, 2025-04-01:
Hi there! I was trying to upgrade the neuropix-PXI plugin to 0.7.X, but wasn't able to add the plug in to the signal chain. The upgrade itself was successful, but every time I tried to add the plug-in to the signal chain, the software would stop working. Please see the attached files for the configuration log. Thank you for your time and help!

[configs-api8.zip](https://github.com/user-attachments/files/19557405/configs-api8.zip)

ziyuw2, 2025-04-01:
Also, the reason why I want to update to 0.7.X is because of the following warming on [this page](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Neuropixels-PXI.html), and I'm recording using neuropixels 2.0 multishank.
"In version 0.6.x of this plugin, commercial Neuropixels 2.0 (multishank) probes did not have the correct scaling applied to account for the lower bit depth of the ADC compared to the beta probes (12 vs. 14 bits). To correct this, any data acquired with from these probes with this plugin should be multiplied by 4. This issue is fixed in version 0.7.x."

And I don't quite understand what it means by "any data acquired with from these probes with this plugin should be multiplied by 4.", so I really want to just upgrade the plug in and not deal with this warning.

anjaldoshi, 2025-04-02:
Hi @ziyuw2,

This issue appears to be a duplicate of the one posted here: https://github.com/open-ephys/plugin-GUI/issues/648.

The crash seems to be caused by a runtime library version mismatch. Could you try installing the latest version of the Microsoft Visual C++ Redistributable from [this link](https://aka.ms/vs/17/release/vc_redist.x64.exe)? Once installed, try using Neuropix-PXI v0.7.1 and check if that resolves the crashes.