open-ephys/plugin-GUI #653: Recording stop: OneBox & Neuropixel 2.0

Daruum, 2025-04-03:
Hi everyone,

I'm currently experiencing a recurring issue with my recordings:

After a while, and seemingly at random, the LFP viewer freezes, and the LED on the OneBox indicating the connection status between the box and the probe changes from blue (recording in progress) to off (no probe connected).

At this point, the Open Ephys interface still indicates that the recording is ongoing (the seconds keep scrolling, etc.), but only ADC data continues to be acquired. The ProbeA LFP viewer remains frozen, while the OneBox ADC LFP viewer continues to scroll—consistent with what I observe when reviewing the recorded files afterward.

I'm using the latest version of the GUI (0.6.7) with a simple recording chain: OneBox > Record Node > LFP viewer.

Daruum, 2025-04-08:
My OneBox is connected directly to a 3.0 USB port on the computer. I'm using the latest version of the Neuropixels plugin (0.7.1)

Regarding the computer (laptop) spec:

Device name MSI
Processor Intel(R) Core(TM) i9-10980HK CPU @ 2.40GHz   3.10 GHz
Installed RAM 32.0 GB (31.8 GB usable)
Device ID F88CE816-DD66-481A-8141-10048612F924
Product ID 00330-53311-76386-AAOEM
System type 64-bit operating system, x64-based processor
Pen and touch No pen or touch input is available for this display

Whether I'm recording to the internal SSD (1 TB, NVMe) or the external SSD (8 TB, SATA), it doesn't make any difference. Recording from all the probe channels (384) or just one leads to the same result.

(There was a mistake in my first message: the probe status LED on the OneBox switches from green to off when the LFP viewer freeze is observed.)

Also, the recording usually works for a long time before crashing—typically 2 to 3 hours.

jsiegle, 2025-04-08:
This likely indicates a dropped connection on the hardware side. Is the probe moving at all when the data stops transmitting? Is there any chance there is a physical disconnection of the data cable or ground wire?

Daruum, 2025-04-14:
It's difficult to say. I'm doing chronic implantations in freely moving animals. Since the problem occurred in both subjects I implanted I don't think it's due to the probe (if the problem is hardware). The only remaining possibility would be the cable connecting the probe to the recording box but it's difficult to check.

Daruum, 2025-04-30:
Update:

After investigation, I found that the headstage may have come into contact with the metallic nozzle of the water bottle, which likely introduced significant noise and caused the Open Ephys GUI to crash during recording.

Daruum, 2025-05-01:
When it happens, the GUI keeps working and the OneBox continues acquiring the signal from 'OneBox-ADC', but the LFP trace from 'ProbeA' freezes. When checking the OneBox, it indicates that the probe is no longer connected (LED off).

(I have now isolated the headstage using tape, and I no longer experience this problem)

closed 2025-06-08 by jsiegle