open-ephys/plugin-GUI #657: Openephys stopped working

BovenE, 2025-05-02:
[image]

I am trying to record with openephys I get the error you can see attached in the image above
I have also attached the required files.

What I am doing is the following: I open openephys and start the acquisition and than I control openephys recording through a TTL send by bpod. Doing so openephys crashes everytime giving the message in image 1. In my signal chain is a Onebox acquisition board, LFP viewer, record control and record node. The openephys version I am currently using is 0.6.7.

Could you please help me how to solve this problem so I record openephys through a TTL send by bpod?

Thank you very much in advance for your time and help.

[activity_2025-05-01_15-29-43.log](https://github.com/user-attachments/files/20010570/activity_2025-05-01_15-29-43.log)

jsiegle, 2025-05-02:
Hi there – the TTL signal to start recording should just be a single short pulse (around 100 ms) if the Record Control plugin is in "Edge toggle" mode, or a pulse that's equal to the duration of the recording interval if you're using "Edge set" mode. Based on the log file, it looks like the plugin is receiving many pulses in quick succession.

Can you remove the Record Control plugin and visualize the Bpod TTL pulses in the LFP Viewer to see how often they are arriving?

BovenE, 2025-05-05:
> Waarschuwing: Deze e-mail is afkomstig van buiten de organisatie. Klik niet op links en open geen bijlagen, tenzij u de afzender herkent en weet dat de inhoud veilig is.
> Caution: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi

Yes that works, thank you !

closed 2025-05-12 by anjaldoshi