open-ephys/plugin-GUI #658: settings.xml not saved

vncntprvst, 2025-05-08:
I'm getting the same issue as #390. 
The settings.xml file is only saved for some recordings but not all, even on sequential recordings in the same session, with the same probe attached. 

OS: Windows 10 Enterprise
Open Ephys GUI version: v0.6.7
Probe: Neuropixels 2.0

medengineer, 2025-05-08:
The settings.xml file should only be generated at the start of a new _acquisition_ (i.e. experiment), not necessarily the start of every new _recording_.

If you only stop recording but not acquisition, then start recording again, the signal chain connections cannot have changed and therefore there is no need to generate a new settings file.

The _parameters_ in the settings.xml file may change between or even during recordings, but it would not be practical to generate a new file every time this happens.

vncntprvst, 2025-05-09:
Just to clarify, here's a simplified view of a recording session:
```
├── LO_02_001
│   ├── data_description.json
│   ├── Record Node 101
│   │   ├── experiment1
│   │   └── settings.xml
│   ├── subject.json
├── LO_02_002
│   ├── data_description.json
│   ├── Record Node 101
│   │   └── experiment1
│   ├── subject.json
├── LO_02_003
│   ├── data_description.json
│   ├── Record Node 101
│   │   └── experiment1
│   ├── subject.json
```
Between each recording, I stopped the recording, and pressed + to increment the recording number, then started the recording again.
However, even though the signal chain hasn't changed, the probe map has: I'm using Neuropixels probe, and each recording is made with a different selection of electrodes and/or shanks. Shouldn't a `settings.xml` file be generated with the updated probe parameters?
Note that for spike sorting purposes, SpikeInterface is expecting a `settings.xml` file for each `Record Node 101` folder. Failure to provide one leads to [this type of error](https://github.com/AllenNeuralDynamics/aind-ephys-pipeline/issues/39).

medengineer, 2025-05-09:
Interesting, indeed either pressing + or renaming the 'main' directory between each recording without stopping acquisition should trigger a new settings file and I am seeing the correct behavior on my end. Are your directory naming settings the same as mine? I assume your data_description.json and subject.json files are being added some time after recording? 

[image]

vncntprvst, 2025-05-09:
That is correct. My directory naming settings are the same. And I add the json files later, just before spike sorting.

medengineer, 2025-05-12:
I would load the original/saved settings.xml using File -> Open in the GUI, apply the electrode configuration via the Neuropixels Interface then save a copy via File -> Save As. You don't need to actually start acquisition/recording. Then you'll only need to modify the date/time manually in each copy of the settings.xml and move it into the correct Record Node folder.

Unfortunately, we don't have any relevant tools since we haven't encountered this issue yet. Another option is to map each recording date/time to a probe JSON file and write a script to generate the corresponding settings.xml files using the original as a template.

vncntprvst, 2025-05-13:
Ok, thanks. That wasn't too bad. 
Since the issue happened twice, I suspect it will recur. I'll close the issue in the meantime and re-open it if needed.   
Thanks

closed 2025-05-13 by vncntprvst