open-ephys/plugin-GUI #664: LFP viewer color by shank

chris-angeloni, 2025-07-11:
It would be nice when using multishank probes with flexible channel mapping capabilities (in my case NPX2.0) to color the LFP Viewer channels by each shank. While they are sorted by shank by default, it is often difficult to tell where one shank ends and the next begins from the activity/channel numbers (NPX2.0 channel number assignments are quite unintuitive to me). This issue is particularly salient when creating custom channel maps, with different numbers of channels on each shank.

I am not as familiar with the Probe Viewer plugin, but from my experience it doesn't look like this is a feature there either, but I will play around with it more.

I am not sure if shank information is passed to the LFP viewer, but using this to define channel colors would be quite helpful.

jsiegle, 2025-07-13:
This is a great suggestion, and should be straightforward to implement since the LFP Viewer already has access to shank indices. We will not get to this before the official version 1.0 release, but will try to add it soon thereafter.

In the Probe Viewer, the color gradient along the left-hand side can be used to visually identify consecutive blocks of channels. However, the channel number often jumps even when consecutive electrodes are selected, so this may not always be useful. Coloring by shank should also be possible in this plugin, and is something we'll want to add in the near future.

anjaldoshi, 2025-07-29:
Color grouping by shank is now supported in the LFP Viewer (added in GUI v1.0.1). To enable it, choose the "By Shank" option in the color grouping dropdown menu. 

[image]

closed 2025-07-29 by anjaldoshi