open-ephys/plugin-GUI #667: GUI v1.0.0 Crashes During Long Recording

kaid0212, 2025-07-24:
**Describe the bug**
The Open Ephys GUI (v1.0.0) crashed at around 20min during a continuous recording session. The crash happened unexpectedly without any user interaction at the time. Everything looks fine, like the CPU GPU usage, buffer, RAM. I'm not sure if this is related to the stability issue in the new version.

**To Reproduce**
I'm doing neuropixels 2 acute recording, and the signal chain looks like this:

[image]
[image]

I didn't find the crash log file. Where should I find it?

**Versions:**
 - OS:
Here is the OS and hardware. 

[image]

 - GUI version: V1.0.0
 - Plugin version: 
 Plugins used during the recording:
1. Neuropixles-PXI: 1.0.2-API10
2. NI-DAQmx: 1.0.1-API10
3. Multiband Integrator: 1.0.0-API10
4. Probe Viewer: 1.0.0-API10
5. Neuropixels CAR: 1.0.0-API10

anjaldoshi, 2025-07-24:
Hi @kaid0212,

We've conducted extensive testing with GUI v1.0.0, recording data from Neuropixels probes over multiple days and across different machines, without encountering any crashes or issues. We've also been using GUI v1.0.0 in our experiments at the Allen Institute for a while now. Based on this, we’re fairly confident that v1.0.0 is more stable and reliable than previous versions.

Regarding the signal chain screenshots you shared: it appears that the Neuropixels-PXI plugin is disabled (the plugin editor is greyed out), which is likely causing the downstream plugins to be disabled as well. Only the NI-DAQmx plugin is active. Could this be because the Neuropixels hardware wasn’t connected when you took the screenshot? Or were you recreating the signal chain on a different machine? Just to confirm: when the crash occurred, was the Neuropixels probe connected and actively displaying data in the GUI?

Log files can be found at the following location on Windows:
`C:\Users\<username>\AppData\Local\Open Ephys\configs-api10`
Note that the AppData folder is hidden by default, so you’ll need to enable the “Show hidden items” option in File Explorer to access it.

[image]

Once you locate the log file(s), please upload them here. If there are no logs in that location, you can enable the built-in console in the Open Ephys GUI by going to View → Show Console. It should display real-time console logs from the GUI and the plugins in use. Please check for any error messages and let us know if you find anything.

Lastly, could you confirm whether the recording crash occurred multiple times or was it a one-time issue?

Thanks!

kaid0212, 2025-07-24:
Hi @anjaldoshi,

Thanks for your quick response! 

The pictures are to show you my signal chain. The Neuropixels were connected and conducting a real recording when the crash occurred. 

I checked the folder, but I can't find the log for that recording session. Also, in the console it's a new session. I can't find anything useful.

The crash has only occurred once to date. I think I can close this issue, and if this occurs again, I'll let you know. 

Thank you!

jsiegle, 2025-07-24:
The absence of a recording log means the crash was triggered by something outside the Open Ephys signal chain, which isn't detectable by the GUI's error handler -- for example, operations inside the Neuropixels API. We have seen this happen with some of the prototype Neuropixels Quad Base probes we have on hand, but never with Neuropixels 2.0, so it's a big strange. Please let us know if it happens again!

kaid0212, 2025-07-24:
> The absence of a recording log means the crash was triggered by something outside the Open Ephys signal chain, which isn't detectable by the GUI's error handler -- for example, operations inside the [...]

Got it! Thank you so much!

kaid0212, 2025-08-01:
HI @jsiegle,

Thank you so much for your quick response! I'll try it and get back to you after doing the test!

jsiegle, 2025-08-05:
We have heard other reports of crashes happening with Pylon viewer running in parallel. So this is unfortunate, but not totally surprising!

closed 2025-08-05 by jsiegle