open-ephys/plugin-GUI #668: V1.0 sync output from OneBox not generated.

danieljdenman, 2025-07-24:
Using GUI v1.0 with a OneBox, the output sync from the SMA1 on the onebox stays high, does not pulse at 1Hz even when set to OUTPUT. ADC stream and Neuropixels stream do not synchronize. The configuration works with v0.6.7

v.1.0 screenshot, with the sync pulse from SMA1 copied to the OneBox ADC input channel 0 (stays high):
[image]

v.0.6.7 screenshot, with the sync pulse from SMA1 copied to the OneBox ADC input channel 0 (1Hz pulse):
[image]

anjaldoshi, 2025-07-24:
Hi @danieljdenman 

Just checking if you're currently using version v1.0.2 of the Neuropixels/OneBox plugin. You can verify it by right clicking the plugin editor’s title bar or checking it in the Plugin Installer. We have resolved some ADC and sync initialization issues since plugin version v1.0.0, so it's worth confirming.

danieljdenman, 2025-07-24:
confirmed, plugin V1.0.2

danieljdenman, 2025-07-24:
FYI with GUI v1.0.0, same PC but switching hardware to a PXIe-1071 with imec card and NI-6133, the sync pulse from the imec card works.

[image]

jsiegle, 2025-07-24:
Can you share the contents of the console after loading a OneBox signal chain in v1.0?

danieljdenman, 2025-07-24:
welp, i went to reproduce it and copy the console, after i unplugged the signal generator...and now we are in business, sync pulse fine and all streams synced. i can't reproduce even when plugging in the signal generator. probably can close the issue for now...

jsiegle, 2025-07-24:
Ok, cool. The sync config code is identical between v1.0 and v0.6.7 plugin versions, so we were confused about what could be causing this. Glad you were able to fix it!

closed 2025-07-24 by jsiegle